riscv-collab/riscv-gnu-toolchain #1774: clang C runtime files missing in elf-llvm toolchains

gilbsgilbs, 2025-09-27:
I maintain the AUR packages for most toolchains provided in this repo. Since 2025.09.16, my CI caught that clang supplied in the riscv32 and riscv64 elf-llvm toolchains fails to link simple C programs because the Clang runtime files are missing:

```console
$ cat main.c
#include <stdio.h>
int main() { printf("ok\n"); return 0; }

$ riscv64-unknown-elf-clang main.c
/opt/riscv64-gnu-toolchain-elf-llvm-bin/bin/riscv64-unknown-elf-ld: cannot find /opt/riscv64-gnu-toolchain-elf-llvm-bin/lib/clang/21/lib/riscv64-unknown-unknown-elf/clang_rt.crtbegin.o: No such file or directory
/opt/riscv64-gnu-toolchain-elf-llvm-bin/bin/riscv64-unknown-elf-ld: cannot find /opt/riscv64-gnu-toolchain-elf-llvm-bin/lib/clang/21/lib/riscv64-unknown-unknown-elf/libclang_rt.builtins.a: No such file or directory
riscv64-unknown-elf-clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

- https://framagit.org/gilbsgilbs/aur-packages/-/jobs/3060679
- https://framagit.org/gilbsgilbs/aur-packages/-/jobs/3060680

This also seems to reproduce in an Ubuntu 24.04 docker container, so it does not seem related to the way I build the package for Archlinux. I can confirm that both `clang_rt.crtbegin.o` and `libclang_rt.builtins.a` are completely missing from the package, I can't do anything from packaging side.

Thanks in advance. I can provide any additional logs or debugging information if needed.

TommyMurphyTM1234, 2025-09-28:
> I maintain the AUR packages for most toolchains provided in this repo. Since 2025.09.16, my CI caught that clang supplied in the riscv32 and riscv64 elf-llvm toolchains fails to link simple C programs because the Clang runtime files are missing:
> ```
> $ cat main.c
> #include <stdio.h>
> int main() { printf("ok\n"); return 0; }
> 
> $ riscv64-unknown-elf-clang main.c
> /opt/riscv64-gnu-toolchain-elf-llvm-bin/bin/riscv64-unknown-elf-ld: cannot find /opt/riscv64-gnu-toolchain-elf-llvm-bin/lib/clang/21/lib/riscv64-unknown-unknown-elf/clang_rt.crtbegin.o: No such file or directory
> /opt/riscv64-gnu-toolchain-elf-llvm-bin/bin/riscv64-unknown-elf-ld: cannot find /opt/riscv64-gnu-toolchain-elf-llvm-bin/lib/clang/21/lib/riscv64-unknown-unknown-elf/libclang_rt.builtins.a: No such file or directory
> riscv64-unknown-elf-clang: error: linker command failed with exit code 1 (use -v to see invocation)
> ```

FWIW I downloaded the last five LLVM bare-metal Ubuntu 24.04 CI builds from here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases

(i.e. the builds for 20250928, 20250920, 20250916, 20250914 and 20250829) and none of them contain the following directory:

- `riscv/lib/clang/2X/lib`

(where `X` is `1` or `0` depending on the build). And none of them can compile/link the simple example above. 

So this issue seems to have been present for a while and not just since 20250916 and doesn't seem to be related to the PR's that I mentioned above. 

When I get a chance I'll go back further and see if there is any previous CI build that does have the runtime support and can compile the simple example.

gilbsgilbs, 2025-09-28:
Thanks to both of you for looking into it.

> (i.e. the builds for 20250928, 20250920, 20250916, 20250914 and 20250829) and none of them contain the following directory…

Might have been a fluke on my end, but I had successful builds on 20250914: e.g. https://framagit.org/gilbsgilbs/aur-packages/-/jobs/3055945 . That's why I said 20250916 introduced the regression. Sorry if that turned out incorrect and made you lose some time.

> can you try building with -DCOMPILER_RT_BUILD_CRT=ON, and see if that fixes your issue? You can either use LLVM_EXTRA_CONFIGURE_FLAGS="-DCOMPILER_RT_BUILD_CRT=ON" make ... when building, or add the flag in Makefile.in to the cmake configuration parameters in stamps/build-llvm-newlib. Please let me know if this resolves your issue.

Thanks. I'm currently ill so can't make any promise, but I'll try to test this tomorrow. FYI the AUR packages just take your packages from GitHub releases so as to avoid recompiling the whole toolchains everytime.

TommyMurphyTM1234, 2025-09-28:
How come the Linux LLVM toolchain doesn't seem to need the likes of `clang_rt.crtbegin.o` and `libclang_rt.builtins.a`?
```
End of search list.
 "/home/user/Downloads/riscv/bin/riscv64-unknown-linux-gnu-ld" --sysroot=/home/user/Downloads/riscv/bin/../sysroot -z relro --hash-style=gnu --eh-frame-hdr -m elf64lriscv -X -pie -dynamic-linker /lib/ld-linux-riscv64-lp64d.so.1 -o a.out /home/user/Downloads/riscv/bin/../sysroot/usr/lib/Scrt1.o /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0/crti.o /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0/crtbeginS.o -L/home/user/Downloads/riscv/bin/../lib/riscv64-unknown-linux-gnu -L/home/user/Downloads/riscv/lib/clang/21/lib/riscv64-unknown-linux-gnu -L/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0 -L/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0/../../../../riscv64-unknown-linux-gnu/lib -L/home/user/Downloads/riscv/bin/../sysroot/lib -L/home/user/Downloads/riscv/bin/../sysroot/usr/lib /tmp/main-7d38e1.o -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0/crtendS.o /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-linux-gnu/15.1.0/crtn.o
```

TommyMurphyTM1234, 2025-09-29:
Sorry - had to hibernate the PC last night and resume this morning so the build hasn't completed yet - 84% of the LLVM/Clang build complete so far.

cmuellner, 2025-09-29:
I don't have an overview of the expected functionality of LLVM ELF toolchains.
Therefore, I think it is better to reach out to the upstream LLVM repo for guidance on that matter.
If necessary, we can adjust build flags in this repo.

cmuellner, 2025-09-29:
> Seems surprising that it looks like this repo's build of the LLVM/Clang bare-metal toolchain can't compile and link a trivial test program for a while (at least 5 CI build cycles back) though? Shouldn't that sort of basic failure have been caught by CI testing? And doesn't the fact that it hasn't been reported until now suggest that maybe nobody is really using that toolchain variant? In which case, why build it?

I'm not sure if "trivial test program" is applicable to a program that uses the C runtime function `printf()` with a baremetal toolchain. But there are some LLVM devs who know these details better, and we may ask them for advice.

topperc, 2025-12-11:
I think adding `--gcc-toolchain=<path to base of toolchain>` to the clang command line for the link step will make it work.